iot-root/garden-of-eden #12: What should button do

iot-root, 2023-07-30:
What should the button do? currently, shuts off lights for one minue

ammaratef45, 2023-07-31:
Suggestions:
  - the Button can be used to put ble into discoverability, this can be useful if we decide to allow interfacing with the gardyn using Bluetooth
  - we can add an api that changes the button action (make it configurable)
  -- turn lights on/off
  -- start/stop water

doubleplusthink, 2023-08-02:
Or have it do multiple things. 
single press - toggle lights
long press - toggle pump
double press - something else

bzzbzzz, 2024-03-29:
> Or have it do multiple things. single press - toggle lights long press - toggle pump double press - something else

maybe make this default, but still keep it configurable and use some pattern for discoverability, as ammaratef45 wrote above ?

petercorrea, 2024-05-02:
Closing issue in favor for https://github.com/iot-root/garden-of-eden/issues/36