iot-root/garden-of-eden #3: power loss recovery

iot-root, 2023-07-30:
Currently system has to be cycled a few times after a power loss.

pushmotion, 2026-08-31:
@iot-root revie #100 and #101 both work on the issues related to power loss & recovering state along with a few other enhancements.

pushmotion, 2026-08-31:
Ran into this on a Gardyn Home (Pi Zero 2) and found two distinct mechanisms behind "has to be cycled a few times". Both have PRs open:

**1. Any service start switched the actuators off — #100.**

`PWMLED(pin)` defaults to `initial_value=0`, so merely *constructing* `Light` or `Pump` writes zero to its pin. `app/__init__.py` imports every sensor blueprint at module level and each `routes.py` instantiates its driver at import, so every start of the REST API turned the lights and pump off. Reproduced directly:

```
light at 65%                        ->  pigs gdc 18 = 6500
sudo systemctl restart garden-api   ->  pigs gdc 18 = 0
```

Nothing errors and no log line appears, so the only visible symptom is that the hardware does not match what you asked for — and power-cycling is a reasonable thing to try in response. Fixed by reading the pin's live duty cycle from pigpiod and passing it as `initial_value`.

**2. The state the recovery path reads was never truthful — #101.**

`restore_actuator_state()` reads `~/.garden_state.json`, but only MQTT-initiated changes ever write it. Schedules run through the `light`/`water` CLIs from cron, and the REST API and physical button drive the drivers directly — all invisible to that file. After a power cut mid-photoperiod it therefore recorded `light_on: false` while the light had been on at 65%, and recovery restored nothing.

#101 polls the real duty cycles every 15s and keeps that file in step, and adds `apply_scheduled_state()`, which replays what the schedule says should be running on each MQTT connect — so recovery no longer depends on the file being correct at the moment power was lost. Verified by forcing the light off, restarting `mqtt.service`, and watching it return to its scheduled 65%.

Pump runs are deliberately *not* auto-resumed: that runs on every connect, and a crash-looping service would re-trigger watering each time, whereas a missed run self-heals at the next scheduled one.