iot-root/garden-of-eden #8: add udev rules for cameras

iot-root, 2023-07-30:
The system shall assign aliases to the cameras. 
- /dev/camera-left  map to /dev/video0 
- /dev/camera-right map to /dev/video1

**User Case**
For easy identification when running scripts or code.

Note: Currently there is a custom bin file which appears to take multiple photos and make one good one. 

```
#!/bin/bash
/usr/bin/fswebcam -d /dev/video0 -r 2500x1900 -S 2 -F 10 right-cam.jpg
/usr/bin/fswebcam -d /dev/video1 -r 2500x1900 -S 2 -F 10 left-cam.jpg
```

iot-root, 2023-07-31:
```
specter@gardyn-01:~ $ udevadm info -a -n /dev/video1 > v1
specter@gardyn-01:~ $ udevadm info -a -n /dev/video0 > v0
specter@gardyn-01:~ $ diff v0 v1
8,9c8,9
<   looking at device '/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/video4linux/video0':
<     KERNEL=="video0"
---
>   looking at device '/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/video4linux/video1':
>     KERNEL=="video1"
16,17c16,17
<   looking at parent device '/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0':
<     KERNELS=="1-1.3:1.0"
---
>   looking at parent device '/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.0':
>     KERNELS=="1-1.4:1.0"
35,36c35,36
<   looking at parent device '/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3':
<     KERNELS=="1-1.3"
---
>   looking at parent device '/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4':
>     KERNELS=="1-1.4"
53,54c53,54
<     ATTRS{devnum}=="3"
<     ATTRS{devpath}=="1.3"
---
>     ATTRS{devnum}=="4"
>     ATTRS{devpath}=="1.4"
66c66
<     ATTRS{urbnum}=="159445"
---
>     ATTRS{urbnum}=="157693"
```

iot-root, 2023-07-31:
We could always prompt user on setup, is this the left or right cam?

bzzbzzz, 2024-03-29:
> We could always prompt user on setup, is this the left or right cam?

Or just name them A and B? Or only an option to switch left/right afterwards?
Because the probability that it's already "right" is >=50%
and even for the cases where it's not, there are quite some people who probably would'nt care ?