iot-root/garden-of-eden #19: standardize logging

iot-root, 2023-07-31:
things to log: 
 - when something is turned on/off
 - errors
 -

ammaratef45, 2023-07-31:
More considerations:
 - start a new log file every (hour?)
 - keep only (3?) older files
 - send the log files to the API before deleting them

iot-root, 2023-07-31:
There is a /usr/bin/logger currently used. Might be a good way to go. 

see `sudo crontab -u root -l`

```* * * * * /usr/local/etc/cronjob_tasks.sh 2>&1 | /usr/bin/logger -t GRD-CRN-SYS```

pushmotion, 2026-09-01:
A concrete item for this from a 2.0.0 code review: `app/sensors/pump/pump.py` uses `print()` for all of its output, where the equivalent `light.py` uses `logging`.

```python
def off(self):
    print("Turning pump off")

def set_duty_cycle(self, duty_cycle_percentage):
    ...
    print(f"Setting pump duty_cycle to {duty_cycle_percentage}%")
```

This matters more than the usual style argument, because of where the pump driver runs:

- Under `mqtt.service`, `print()` goes to stdout → the journal, not `mqtt.log`, so pump activity is missing from the log file everything else writes to.
- It bypasses `LOG_LEVEL` entirely — there is no way to quiet it or raise it.
- It never reaches a `logging` handler, so anything that mirrors log records elsewhere (we forward `WARNING+` to an MQTT topic for a "Last Log" diagnostic sensor downstream) can never see pump events.

Swapping the six `print()` calls for `logging` calls matching `light.py`'s style is a small change and makes the pump — the one actuator with a hard safety cap — as observable as the light already is.

Two adjacent nits in the same file, if anyone is in there:

- `set_duty_cycle()` raises `ValueError("Speed must be between 0 and 1")` on out-of-range input, but the accepted range is 0–100.
- `on()`'s docstring says "Turn pump on. Default to 30 percent duty" while the body sets `self.pump.value = 1` (100%).