iot-root/garden-of-eden #36: button interface

iot-root, 2023-08-13:
**Requirements**

TBD

The button shall have:
- [ ] a button press monitor
- [ ] it shall trigger actions?

Notes:
- [GPIO-13 pin description](https://pinout.xyz/pinout/pin33_gpio13/)

iot-root, 2023-11-17:
@ammaratef45  Thoughts on the following : 

The button shall cancel the pump and light if pressed once.

Future add - maybe we add wifi provisioning reset on hold for 5 seconds etc.

ammaratef45, 2023-11-18:
@iot-root I also think it would be a good idea to make the button action changeable through API

iot-root, 2024-01-03:
I was thinking we need to add a wifi provisioner similar to the functionality now otherwise you need to tear the entire head unit open in order to reconfigure if wifi is ever deauthorized.

iot-root, 2024-01-03:
could do something like this https://github.com/ahepp/wlanwd/blob/master/README.md

iot-root, 2024-04-27:
Currently using this on my gardyn and need to clean up and document:
https://github.com/iot-root/python-wifi-recovery