iot-root/garden-of-eden #46: how is this used?  I assume SSH access is required to load this and test?

x0SiN0x, 2023-10-12:
It's not clear to me how this is intended to be used without being able to access the gardyn host.  Is the username and password documented anywhere to access through ssh?

ammaratef45, 2023-10-13:
You can change the OS on the pi device and ssh into the new OS to do the installation

You should keep a backup of the old OS but this repo is to use as a replacement and not alongside the original FW

iot-root, 2023-10-13:
Yes, you will need the ssh username and password or you can pull the SD card. I found adding a pi zero 2 instead of first gen was significant in development and performance.  If you message me on https://discord.gg/H35VwazSDb , I will send you the username and password. I am not sure the legality of posting it, hence why it is not posted on this repo.

The project is in progress and it took a fair amount of work to reverse engineer and get this far. So if there are gaps feel free to join us on https://discord.gg/H35VwazSDb or submit ideas/changes we are looking for collaborators.

x0SiN0x, 2023-10-13:
ah thanks for the clarification - I will be ordering a pi zero 2 and trying this out.

closed 2023-10-13 by x0SiN0x

iot-root, 2023-10-13:
@x0SiN0x let me know if you need any help. I sent you everything on discord.

iot-root, 2023-10-13:
@x0SiN0x make sure you buy a pi zero 2 `WITHOUT headers`