iot-root/garden-of-eden #51: fix readme and setup script

iot-root, 2024-04-07:
- Ensure i2c is confgired
- run gpiod 
- setup systemctl to use the user and project path
- udev rules for camera aliases which need 
  - Note: vendor and product IDs are different for gardyn 2.0 vs 3.0
- included basic HASS automation config in Wiki and broker instructions (already have a broker workflow?)
- using bin/api-test.sh to determine if everything is working....
- mosquitto_pub -h localhost -t testtopic -m "Hello, MQTT!" -u "yourusername" -P "yourpassword"

petercorrea, 2024-04-22:
@iot-root can you update this and split out anything that could be separated? I'm combing the issues and relabeling so we can better tackle things.

iot-root, 2024-04-27:
yes, let me split the readme from setup script

iot-root, 2024-04-27:
I moved out camera and readme

iot-root, 2024-04-28:
Fixes per bbf006c