iot-root/garden-of-eden #63: We will need some ideas for the home assistant dashboard cards.

petercorrea, 2024-05-01:


bzzzbzzzzz, 2024-05-08:
https://github.com/iot-root/garden-of-eden/assets/166843795/ee9352bf-b6bb-46a5-bbe9-1852367b8658

?

pushmotion, 2026-08-31:
A couple of card-able entities coming out of open PRs, both added because the existing dashboard showed something misleading:

**Next Pump Run** (#102) — a `timestamp` sensor, so HA renders it as a relative time ("in 2 hours"). The existing *Pump Run Time* entity is fed by `_everyday_pump()`, which can only express a single run per day; on a multi-cycle schedule it shows the first run and never moves. A tower watering seven times a day sat frozen on 1:00 AM with nothing indicating the next run. Deliberately a separate read-only sensor, because that entity is writable and editing it rewrites every day down to the one run it can express.

**Water Depth / Water Remaining / Water Gallons** (#95) — derived from calibrated tank geometry so they rise as you fill, instead of the raw sensor-to-surface airgap that grows as the tank drains. Details in #86.

Also worth knowing when designing cards: before #101, the light and pump state topics only reflected changes the MQTT service itself had commanded, so a dashboard would show the light OFF while it was on at 65% and OFF through an entire scheduled pump run. Any card built on those topics was showing fiction whenever cron, the REST API or the physical button made the change.