iot-root/garden-of-eden #78: Add the gardyn button as a home assistant button

thoughtleader, 2024-07-30:
Be nice if I could use the silver button to acknowledge tasks like when I add fertilizer or additive to the water. I have a screen near the gardyn I can pop notifications on, so if I could use that button id be so happy. All of that automation I can handle but it would be nice if someone could have that button publish press events to mqtt

iot-root, 2024-09-06:
> Be nice if I could use the silver button to acknowledge tasks like when I add fertilizer or additive to the water. I have a screen near the gardyn I can pop notifications on, so if I could use that [...]

Feeling out your needs here. Would you prefer the button do it or that we add a seperate isolated piece of hardware that could do that like an esp32 with a LCD of stats and some buttons?

thoughtleader, 2024-09-26:
I was just thinking that enabling the button to register events in hassio would provide some additional functionality like i was summarizing above. I have a nice panel in Hassio, and I also have a NSPanel nearby, that I'm tracking dates of certain events, like fertilizer and water changes, so having additional hardware to display info seems like overkill.

iot-root, 2024-09-27:
> I was just thinking that enabling the button to register events in hassio would provide some additional functionality like i was summarizing above. I have a nice panel in Hassio, and I also have a [...]

I was thinking about this and maybe adding a m5stack solution like https://www.adafruit.com/product/4289 

Background:
I messed around with various button combinations , press, double press, press and hold and it can be a little finicky to make this all reliable. That said I will see what I can do and if you would like to brainstorm further let me know. 

You can see this repo where I am working on AP wifi recovery: https://github.com/iot-root/python-wifi-recovery/blob/main/scripts/gpio_monitor.py